lucysrausch/hoverboard-firmware-hack #110: PPM mode little unstable in neutral

franck24300, 2019-06-24:
Does anyone know why in PPM RC mode, the wheels are a little unstable when the joystick is in neutral? This does not happen in PWM RC mode of benjaf.

snowkingcro, 2019-06-24:
Benjaf's PWM mode works flawles, soft start and excelent responds to acceleration, when in neutral, the wheels are in brake mode and are hard to turn... Excelent repo, except it has no steering, than it would be flawles...

franck24300, 2019-06-24:
I already know that, but that does not tell me why the Niklas PPM mode or other derivative is a bit unstable in neutral.

Jan--Henrik, 2019-06-24:
PPM is in general quite unstable. Make sure that the cables are as short as possible and you can add a filter capacitor to the input to reduce noise

benjaf, 2019-06-24:
The standard PPM code does not include deadband. Since the input from the receiver is rarely a true zero, you will see some jittering. 
See line 125 in my version of control.c for a needlessly complicated example of how it can be handled.

franck24300, 2019-06-24:
@Jan--Henrik   I already did that and it's the same. We should add a dead zone as did benjaf.

verial, 2019-08-22:
peut votre config.h et main.h ?

verial, 2019-08-22:
can I have your config.h and main.h ?

franck24300, 2019-08-22:
@verial 
Here is my config.h
I did not change the main.h it's Niklas's original.

[config.zip](https://github.com/NiklasFauth/hoverboard-firmware-hack/files/3528817/config.zip)

verial, 2019-08-22:
since you do not work for me.

verial, 2019-08-22:
I want to try PWM.

vfear, 2019-08-22:
@benjaf Hello, I downloaded your version of the firmware (benjaf pwm). When I want to stop the wheels - they rotate freely, without the brake. Why is this happening? How to make the brake stronger when the wheels are standing?
And another question. How does the child mode turn on? Need some wires to connect with each other?

benjaf, 2019-08-22:
@vfear My version does not lock the wheels when the throttle is in neutral. This behavior can be changed by modifying this section of [main.c](https://github.com/benjaf/hoverboard-firmware-hack-esk8/blob/9aca8c7939ec418b41ac4995fb3cfb2911dfd727/Src/main.c#L282)
Child mode is enabled by connecting the blue wire of the right sensor board (PB10) to ground.

vfear, 2019-08-22:
@benjaf Thanks, I did as you said - it's all right now) But the brakes are not strong. The wheel can be easily twisted by hand. In firmware (NiklasFauth UART), the wheel hold is much stronger. Is it possible to change this?

vfear, 2019-08-22:
@benjaf And wanted to know if the steering feature would be added in the future?

benjaf, 2019-08-22:
@vfear I don't know why the hold power would be different in my version, I haven't changed the way power is applied to the wheels. It might behave differently because speed is set to exactly 0 when throttle is neutral, as opposed to 'pretty much zero' which is what you get when there is no deadband.
Steering might be added at some point, but at the moment I don't have any hardware which can use it.

vfear, 2019-08-22:
@benjaf Thank you for your excellent work =) I Will wait very much when the steering appears. And the dead zone changes here (#define PWM_DEADBAND 100)?

benjaf, 2019-08-22:
@verial Yes, that is correct. Setting PWM_DEADBAND to 0 will disable it.

benjaf, 2019-08-23:
@vfear Have a look at the new branch of my code, steering has been added.

verial, 2019-08-25:
where can I see?

vfear, 2019-08-26:
@benjaf Thanks. Where do you get the power supply for the receiver? 5v

snowkingcro, 2019-08-26:
@vfear you can take it from the red wire from hall. (Not the red sensor).

vfear, 2019-08-26:
@snowkingcro Receiver FS-r6b. Connected the red wire from the hall, and the ground to the battery. After 15 minutes it broke down and heated up strongly =)

snowkingcro, 2019-08-26:
Did you measure the output voltage from those two pins?

benjaf, 2019-08-26:
@vfear I have not had this problem. I have had the receiver connected to +5V and GND of the hall wires for quite a long time now with no issues.

vfear, 2019-08-26:
@snowkingcro Yes, everything is OK
@benjaf I think maybe the problem is that the receiver I have is another signal wire to the arduino... which controls the motor steering, children's cars.

benjaf, 2019-08-27:
So... Based on the input pulse on pin 3, either pin 7 or 8 is set high. Correct?
And the deadzone is 5 in either direction. Values above center enables pin 7, and values below center enables pin 8.
And the pwm value in either < 1500 or > 1500 ranges is always assigned to pin 9.
It does some sort of tank steer, with one channel pr. wheel, I assume?

The delay needs to be removed. PulseIn waits until a pulse is registered and a delay of 20ms will cause the code to skip a lot of pulses.

Other than that, I don't see why it would not work. I would probably do the map() only once per range and store the result in a variable, that would make the code a lot easier to read and debug.

vfear, 2019-08-27:
@benjaf Hello, help please. Doing the steering of my child's car. I want to do the steering, as you have in the program with the dead zone PWM. Steering turn BTS7960 motor driver and Arduino. Can you tell me how your PWM code works?
`//FlySky CT6B

int CH2;

int n2;
int S1;
void setup()
{

pinMode(7, OUTPUT); // Left
pinMode(8, OUTPUT); // Right

pinMode( 3 ,INPUT_PULLUP); 
}
void loop()
{

analogWrite(9, S1); //PWM
 
 CH2  = pulseIn(3, HIGH);

if(!((CH2) == (0)))
{n2=CH2;}
else
{n2=1500;}

digitalWrite(8, ((map((n2), (1100), (1500), (255), (0)))) > (5));
if (((map((n2), (1500), (1900), (0), (255)))) > (5)) {
S1 = (map((n2), (1500), (1900), (0), (255)));
}

digitalWrite(7, ((map((n2), (1500), (1900), (0), (255)))) > (5));
if (( (!(((map((n2), (1100), (1500), (255), (0)))) > (5))) && (!(((map((n2), (1500), (1900), (0), (255)))) > (5))) )) {
S1 = 0;
}
if (((map((n2), (1100), (1500), (255), (0)))) > (5)) {
S1 = (map((n2), (1100), (1500), (255), (0)));
}

delay(20);

}`
Here's an example of my code. But it works very badly. Want to make it like you have in the firmware. Help please =)

benjaf, 2019-08-27:
@vfear I'd probably use PULLDOWN in stead, but I don't think it will make much of a difference in this case.
A timeout on the pulseIn() would be a good idea too, since it should never wait for more than ~30ms.
And by the way, your values might be off - afaik, the pulseIn() returns pulse width in microseconds, which would make the desired range of 10-20ms 10,000 - 20,000, not 1000-2000.

vfear, 2019-08-27:
@benjaf Thanks. The data from my receiver is very unstable. I wanted to know how you fix it. I have a very bad work.

benjaf, 2019-08-27:
@vfear never mind, I was looking at the wrong numbers. It could be that there is to much noise on the lines. How long are your signal wires?

vfear, 2019-08-27:
@benjaf pinMode( 3 ,INPUT_PULLUP); - did I do it right?

benjaf, 2019-08-27:
@vfear You will probably need to debug via serial console then, just to see what the values read by the arduino actually are.

vfear, 2019-08-27:
@benjaf Can you give an example? I don't understand a little =)

vfear, 2019-08-27:
@benjaf 3-5 cm

vfear, 2019-08-30:
@benjaf Hello, wanted to know how to enable the brake (Soft braking)? Green wire with GND?

benjaf, 2019-08-30:
@vfear The soft braking is enabled by default whenever the speed is being decreased. You can change the value of BRAKE_FILTER in config.h to make it softer.

vfear, 2019-08-30:
@benjaf Is it possible to do braking with a button?
now by default when you release the joystick - disabled motor

benjaf, 2019-08-31:
@vfear You could use a button, but disabling [this](https://github.com/benjaf/hoverboard-firmware-hack-esk8/blob/bbd6c69ae4286ba26ba38a161dfd33c3a254d07c/Src/main.c#L282) section of code would keep the motors enabled always.

samurairai, 2020-04-28:
> 
> 
> @vfear You will probably need to debug via serial console then, just to see what the values read by the arduino actually are.

May I know the pin connection for the console?  I cant find any info on it :(