lucysrausch/hoverboard-firmware-hack #111: PWM RC with 2 canals

franck24300, 2019-06-25:
Apparently this Russian has managed to operate the direction using a RC PWM receiver, but he does not say how he did:

benjaf, 2019-06-25:
I might revisit full RC PWM control at some point, but at the moment I am awaiting a pull request on the bipropellant fork before moving any further.
Anyone wanting to implement PWM control can look at my fork for inspiration. I suspect that the timer I used for the steering channel was also being used elsewhere, but I did not investigate further.

snowkingcro, 2019-06-30:
That would be genious... And very generous from you too 😎

virus81-pro, 2019-08-05:
That would be genious... And very generous from you too

benjaf, 2019-08-23:
All right - I have created a new branch on my fork, which has full control via RC PWM.
Having the channels share a single timer made all the difference, so there it is!

verial, 2019-08-25:
where can I see?

benjaf, 2019-08-25:
@verial Here: [RC Steering](https://github.com/benjaf/hoverboard-firmware-hack-esk8/tree/RC-Steering?files=1)

verial, 2019-08-28:
It's not working

verial, 2019-08-28:
> @verial здесь: [рулевое управление RC](https://github.com/benjaf/hoverboard-firmware-hack-esk8/tree/RC-Steering?files=1)

не работает

snowkingcro, 2019-09-01:
@benjaf  Just to report that it's finaly working! Smooth and precisely! Thank you for your time and will for us noobs...

verial, 2019-09-02:
https://www.youtube.com/watch?v=Xee27lz3rug&feature=youtu.be

lukis01, 2021-04-29:
Sorry to bother you but I tried several attempts to wire and pair RC transmitter and receiver from AliExpress but it doesn't work. I used benjaf's forked files and got working pwm on flysky gt2b (ch2-ground,positive,signal and ch1-only signal). Everything worked,until I decided to change to different RC kit-after connecting both wheels act like crazy (spinning without control both ways). Could someone help me please? That flysky kit works fine,but I'd prefer much smaller transmitter for my project..

Candas1, 2021-04-29:
Are you sure this RC kit is doing PWM ?
PWM protocol is so poor that it will pick anything as a signal.
For example if you send PPM or serial/USART it can go crazy.

lukis01, 2021-04-29:
well...before you replied, I thought that all RC kits are using PWM :D  I "just" flashed PWM version from benjaf's repo using some custom changes in config file, connected flysky et voila...everything worked .... Now, after some google diggin I am not so sure, that this little piece [s](https://i.ebayimg.com/images/g/yJ0AAOSwW-9fteVL/s-l1600.jpg) is using the same type of signal...thank you for pointing me to a possible solution..

benjaf, 2021-04-29:
That small RC kit looks just like the one I'm using, so it definitely should work.
Do you have any servos around to check that the receiver actually outputs a valid PWM signal?

lukis01, 2021-04-30:
yes, I bought some SG90 servos just to be sure that kit is OK and they both worked perfectly (I wasn't sure if kit is paired properly). Yesterday I found some additional information about the kit - it isn't supposed to work under FOC. Couldn't be a problem here? I am using control mode 2 (SPEED) with reduced values (spped+steer coefficient)..

benjaf, 2021-05-01:
I honestly have no idea why it wouldn't work if the receiver is connected correctly and outputs a proper PWM signal. The protocol is very simple, and doesn't really leave much room for interpretation!

lukis01, 2021-05-01:
[image]

lukis01, 2021-05-01:
I am using FOC mode in config file (value 2) together with this receiver wiring (ch2 for throttle and ch1 for steering).same wiring and adjustment I use for flysky that works properly...could you please check your mode settings just to be sure please?

Candas1, 2021-05-01:
So you are not using Niklas firmware but Emanuel's firmware?

snowkingcro, 2021-05-01:
The proof that benjafs firmware works is here: https://youtu.be/d5Pv82Sy5zc

I flashed my board with his repo /steering and use a common 3ch RC car receiver/transmitter.

Candas1, 2021-05-01:
Yeah but I am not sure what firmware we are talking about anymore... I assume people raise issues in the repository they are using.

benjaf, 2021-05-01:
It looks like we are taking about Emanuels firmware, possibly my fork of it (which is way behind the main branch and should not be used!).
If that is the case, any issues should not be discussed here but rather on the appropriate repos.

Candas1, 2021-05-01:
On Emanuel's fork pwm variant is using the right sensor cable by default because it's shorter, and the left one for debug usart. But lukis01 didn't mention if he used different forks

mrlunk, 2021-05-09:
Works fine for me here's the proof:

https://www.youtube.com/watch?v=P9L5MrHEN-k

I'm using left sensorcable (shortened).

PL.

birdemirleblebi, 2023-03-05:
Hello, I have the same problem as above. (spinning without control both ways)

The software I use: https://github.com/benjaf/hoverboard-firmware-hack-esk8/tree/RC-Steering

Here is the controller I use: https://www.hizliresim.com/ackcrkb

Do you have a suggestion?

Candas1, 2023-03-06:
I suggest you read this thread... It says use Emanuel Feru's firmware and post your question in the right repository...