lucysrausch/hoverboard-firmware-hack #113: free online hoverboard firmware compiler :-)

RoboDurden, 2019-06-27:
### -> https://youtu.be/15UBBcZvMh8

maybe you want to add more default settings to conifg.h like

`// ###### SIMPLE BOBBYCAR ######`
`// for better bobbycar code see: https://github.com/larsmm/hoverboard-firmware-hack-bbcar`
`// #define FILTER              0.1`
`// #define SPEED_COEFFICIENT   -1`
`// #define STEER_COEFFICIENT   0`

and

`// ###### ARMCHAIR ######`
`// #define FILTER              0.05`
`// #define SPEED_COEFFICIENT   0.5`
`// #define STEER_COEFFICIENT   -0.2`

My online compiler searches for `// #define FILTER` and then modifies all the define constants listed till the next empty line. For the description for such a default settings, my parser looks at `// comment` in the two previous lines.

ideas welcome
Roland
the little physicist :-)

Jan--Henrik, 2019-08-02:
Hey,

I just saw your awesome online compiler, very cool! I also added it to the README (if that is okay with you). :)
Nevertheless, we wouldn't like to put more standard configurations in config.h as this is rather confusing. If someone likes, they can clone the repo of course and do this so you can add the repo to your online compiler.

RoboDurden, 2019-08-03:
It is honor to be mentioned on your root repo :-)

Yes i too kept only to your two basic BOBBYCAR and ARMCHAIR presets with my reliable-uart fork https://github.com/RoboDurden/hoverboard-firmware-hack .
These two settings are a good start for beginners.

### Maybe you would like to take notice of my object oriented rebuild: 

https://github.com/RoboDurden/oo-hoverboard-firmware

At least it already compiles as c++ code :-)

And i dream to rewrite it as a full operating system for these nice,cheap generic hoverboard controllers. 

One of the greatest inventions of the last century was not the car nor the atomic bomb but object oriented programming. And already more than a decade ago i (physicist) noticed what will be one of the greatest inventions of this century: I named it OON = object oriented networking: You simply throw all the OON-objects into the boiling pot and they do the networking themselves. 

These hoverboard controllers might become a great OON-OS for moving objects, robots, windmills, etc. etc.

from https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/59

> frustration is part of learning :)

### No definitely NOT !

This again confirms my strong dislike for the linux community who still have not heard of oop.
I would try to replace the gcc with g++ in the Makefile.
Then an abstract class CModule and and array CModule aModule[10]
And all the different Controls in their own `//#include "CModuleAdc.h"` which simply push themselves onto aModule.
And in the main.c simply loops like 
`for (int i=0; i<iCountModule; i++) aModule[i].Init();`
and
`for (int i=0; i<iCountModule; i++) aModule[i].Tick();`

Then you could easily add your new CControlUSART and CReadSpeed to the originial repo ..

That would be joy !

...

all the #ifdef CONTROL_ throughout the source code is simply horrific. And especially for a github project where multiple people want to contribute, the first fork of Niklas his repo should have been a c++ port.

...

> How do you plan to get rid of all the #defines without generic communication protocol?

Every CModulControl will implement its own communication "protocol". The CControlAdc will read the two analog inputs, the CControlUART will have its communiction strct, the CCControlTest will have no input :-)
All CControl classes will return a command-struc (steer,speed,etc) when CControl::Tick() is called.

A bit more tricky will be CControlUART to access CControlSensor because it needs the actual wheel speed, battery voltage, etc. to send back via its private UART protocol.
The CControlSensor could be a "friend" of CControlUART.
Or every CModule will have access to the global aModule and can iterate through all the other modules to find the ones they would like to retreive data from or send data to.
So of course there could be a CModuleBldc and every CControl can send directly the speed and steer to it. Or there could be a module CModuleMain which first retrieves the speed/steer from a CControl and then sends the final speed/steer to CBldc :-)
And every CModule in its own .cpp file and the main.c would only need the  wanted `#includes`. 

Then we could all return to the original NiklasFauth repo and there would be all kinds of modules being maintained by their creators.

...............................................................

This already hints for OON where each object from all the different classes communicate their skills and learn how to act as one big solution..

Am already happy that the code compiles but can not continue for the next weeks. I now am on my 41th 500 im trip in northern Bavaria and have no spare control board with me: https://youtu.be/LHfGknwLSQw

das Roland
the little physicist :-)

RoboDurden, 2019-08-26:
The new firmware seems to behave a little different. Try playing with the power on switch. Always on, always off, short on pulse. And unplug and plug-in the stlink usb. Also make sure you have 36 volt. I too feared a few times I had bricked the board.

lolex3788, 2020-08-11:
I'm make an rc shopping cart with a hoverboard and ready to install the firmware.

Is it only to click on ppm control on the compiling site and send it to the motherboard or do I need to do something ells first?

RoboDurden, 2020-08-11:
Yes.
Of course you can do lots of thing wrong.

lolex3788, 2020-08-11:
I don't really understand what you mean with that answer or how its helpful but okay.

lolex3788, 2020-08-11:
Like what DO I have to do on pioneerland.de/hoverhack/ to get a firmware that works with ppm?

vamfun, 2021-01-05:
I thank you for giving us a beginner compiler... but of course I quickly found I have a board with the buzzer and motor voltage on different pins.   If I had access to defines.h and setup.c I could have continued... but had to use Keil to fix my code.   Any chance your compiler can expose more of the files...or did I miss something?

RoboDurden, 2021-01-05:
Of course I could add more source files.
But in your case it would be better if you fork the Emanuel Feru firmware and I add your repo so that such boards with different pins become available to all of us.

It is so depressing to see everyone only care for himself :-(

I would like to stick to only config.h and main.c to allow some basic adaptions.
Adding more source files will end up in a complete online ide. That would be foolish!

RoboDurden, 2021-01-07:
> If I had access to defines.h and setup.c I could have continued... 

But yes, if you can link me an opensource javascript ide with a white list of .c and .h files, where i only need to add a dropdown to set the project root to temporary copies of my cloned repos, i'll be happy to add it to my online compiler :-)
I just don't want to code just another ide even so it would only take me a few hours.

vamfun, 2021-01-12:
Understand...I wasn’t sure what was required to source these files.   I probably can’t do what you asked since my java code knowledge is  limited.  
Thanks for your reply...I will however think about the idea.
Danka 
Chris

http://vamfun.wordpress.com
Robot musing blog
iPhone

RoboDurden, 2021-01-21:
Okay i have spent a week to code my own simple IDE: https://github.com/RoboDurden/MinIDE

And based my new online compiler on it: **https://pionierland.de/hoverhack/**

> ... If I had access to defines.h and setup.c I could have continued.

If you like it, send me 5 Euro: https://www.paypal.com/donate?hosted_button_id=L33LZL5PDDYCL :-)

Ideas welcome :-)