lucysrausch/hoverboard-firmware-hack #115: Cant flash whitout 3.3v conected to stlink...

DaviVolcan, 2019-07-14:
When i try to flash the board using stlink v2 io, clk, gnd...i have no luck...(im following even the Power button instructions) Only when i plug the vcc from stlink to 3.3v on main board  I get conection and finally i can flash... Does someone knows why? 
the readme.md says that connecting vcc from stlink to 3.3 could kill the main board... Why? 

Thanks!

Combinacijus, 2019-07-15:
I have heard that if you power stm32 microcontroller externally and also power from stlink you will burn it. Don't know if it's true but I don't wanna find out

p-h-a-i-l, 2019-07-15:
@DaviVolcan Are you using an [original stlink/V2](https://www.st.com/resource/en/user_manual/dm00026748.pdf)? If I recall correctly, the need to be connected to 3.3V to sense the VCC voltage level.

larsmm, 2019-07-15:
You do not want to have 2 3.3V sources: from stlink and from battery. If they are a bit different, current will flow where it is not very healthy.

If you power the mailboard from stlink only I do not now if the 3.3V regulator likes it to be powered from the wrong side.

p-h-a-i-l, 2019-07-15:
see https://www.st.com/content/ccc/resource/technical/document/user_manual/65/e0/44/72/9e/34/41/8d/DM00026748.pdf/files/DM00026748.pdf/jcr:content/translations/en.DM00026748.pdf

[image]

p-h-a-i-l, 2019-07-15:
You could also test my theory by keeping the board save. Either with a 100k resistor in series or an independent 3.3V source.

alex-makarov, 2019-08-01:
I *probably* have the same issue as @DaviVolcan. ST-Link V2 clone connected to the correct pins, without 3.3V - as per instructions (I've checked that the connection is indeed correct, by testing directly on the pins of the MC). However neither st-link nor openocd can read anything, tried on two PCs.
 
The microcontroller (gd32f103 in my case) seems to be alive -- the board was new and functioning hoverboard; after disassembly it beeps two times on power-on and the onboard LED is blinking. I would want to try supplying the 3.3V but really don't want to kill the board - how exactly can I try it in safer way, @p-h-a-i-l ?

DSBRobotics, 2019-08-01:
Measure the 3.3v on board and check ground.

alex-makarov, 2019-08-01:
> Measure the 3.3v on board and check ground.

Ground OK, 3.3v looking good on the SWD programming header.

alex-makarov, 2019-08-03:
Maybe any other ideas how to move further from the point of not being able to flash? What else to check? Thanks!

alex-makarov, 2019-08-03:
OK, solved it! The problem was with ST-LINK firmware, apparently the programmer (cheap chinese clone of ST-Link v2) was shipped with some broken firmware. After I've updated it using ST Windows utility running in VirtualBox, everything worked.

timsooonnnn, 2019-08-25:
this youtuber sells your hoverboard program for 1000. Isn't your program code free?
https://youtu.be/LOpbEYA0hio

verial, 2019-08-26:
I'm flashing from St-link source.

closed 2020-03-20 by Jan--Henrik