lucysrausch/hoverboard-firmware-hack #122: Mainboard dead after flashing?!

snowkingcro, 2019-09-17:
So, i flashed my firmware into the board from the pic below. (The same firmware works on my other board).
I always flash the way described in the repo.
So, after flashing and removing the jumper, the board shuts down.
It gets on by returning the jumper and stays on until jumper removed.
I don't get the beeping sound for startup and shut down from speaker.
When charger is connected, the board gets on (without sound) and remain on...
The board doesn't react on any input... 
Is there a way to troubleshoot it? Or is it possible that this board is not suported?
[image]

bs0d1, 2019-09-30:
Hi it sounds like you just deleted the Firmware that was on it. The Beeping is done in The Firmware not the Hardware. It is very inprobable that your board is dead. Plesse just try to flash something onto it.

Chaos99, 2020-01-16:
This is another board than the original one. Some signals use different pins on the controller. this includes the power control and the buzzer, this is way it seems dead. See this thread for a solution:

https://github.com/bipropellant/bipropellant-hoverboard-firmware/issues/90

closed 2020-03-20 by Jan--Henrik