lucysrausch/hoverboard-firmware-hack #127: 10" Hoverboard

jpiat, 2019-10-11:
Hi, 

apart from the wheel size, is there any other modification to make to config.h ? On my 10" hoverboard, the left wheel has issues achieving the target speed while the other wheel works nice. Is there any filter parameter that would need to be adapted for 10" wheels ?

benjaf, 2019-10-11:
Internally the 10" boards are exactly the same, the only difference is the external "shell" of the wheel.
It could be a bad connection in one of the sensor wires?

jpiat, 2019-10-14:
It seems that i either have a hall sensor problem or one of the motor phase is dead. I'll try replacing the control board.

jpiat, 2019-10-15:
I replaced the control board but the problems remains. I discovered that when driving backward (negative speed). The right wheel becomes the problem and the left wheel works fine. Could it be that the HALL sensor layout is different in the 10" inflatable wheels ?

EmanuelFeru, 2019-10-16:
@jpiat  I would doubt that the Hall sensor or the motor for the 10" wheel is different because they sell those after-market control boards and they claim it works for both 10" and 6.5". 

I would suspect a wrong Hall wire or Phase wire mapping to this firmware itself. If you have an oscilloscope (with minimum 2 channels) you could check the mapping by looking if the Hall pulse is aligned with the respective Phase back-EMF as shown in figure below (this is what the firmware expects). You can do this experiment by spinning the motor by hand. Of course, if the spinning direction is reverse than the peaks of the sine back-EMF are on the bottom.
[image]

If you don't have an oscilloscope, you could also try the hard way, by swapping Phases or hall wires. And carefully applying low PWM values.

jpiat, 2019-10-16:
Hi, thanks for the input. How would that explain that the wheel is fine when driving backward (and the other wheel becomes the issue) ?

p-h-a-i-l, 2019-10-16:
Yes, that would explain the difference in behavior based on rotation direction.

jpiat, 2019-10-16:
Hi, I have found the correct hall sensor cable ordering. I haven't fully tested, but it seems to solve the issue. Here is a picture of what is the hall sensor cable ordering on my board (different from the one pictured on this GitHub).]
[image]

EmanuelFeru, 2019-10-16:
@jpiat nice, good that you found it! The [YST board](https://raw.githubusercontent.com/cloidnerux/hoverboard-firmware-hack/master/pinout.png) apparently also have different Hall wires color mapping.

closed 2020-03-20 by Jan--Henrik