lucysrausch/hoverboard-firmware-hack #128: Controlling the two motors independently

blubbi321, 2019-10-19:
I have a robot on which I'll need the two connected wheels to turn
* in the same way
* in opposing directions
* each of the above forward and backwards for each motor

Best would be to have a really dumb controller (to which I just provide a value per motor in which the sign determines the motor's direction and the absolute value determines the speed) .. So I tried to get rid of all the speed/steering coefficents and in main.c replaced

```
speedR = CLAMP(speed * SPEED_COEFFICIENT -  steer * STEER_COEFFICIENT, -1000, 1000);
speedL = CLAMP(speed * SPEED_COEFFICIENT +  steer * STEER_COEFFICIENT, -1000, 1000);
```

with

```
speedR = CLAMP(speed, -1000, 1000);
speedL = CLAMP(steer, -1000, 1000);
```

expecting that I could now use the speed signal to control the right and the steer signal to control the left motor. 

However, when I turn things on I still get both wheels turning, even when only providing a speed value (via UART).

Any pointers anybody?

EmanuelFeru, 2019-10-19:
To me what you did is correct to make them completely independend. Did you check that cmd1 and cmd2 are indeed different? And also speedR and speedL as well different? I assume some issue with reading the serial command.

blubbi321, 2019-10-19:
@EmanuelFeru thanks for your reply, will double check via the uart output ..

blubbi321, 2020-01-22:
@EmanuelFeru sorry for the long delay. So I checked and indeed the values for cmd1 and cmd2 are the same. What I send in is produced by the arduino code sample from the config file, ie

```
Serial.write((uint8_t *)&steer_value, sizeof(steer_value));
Serial.write((uint8_t *)&speed_value, sizeof(speed_value));
```

Writing 1 for the steer value results in this on UART:

[image]

Which results in 1 == cmd1 == cmd2

Any ideas?

EmanuelFeru, 2020-01-23:
That is strange...  I also implemented serial commands in my repo and it worked. The thing I did differently is that I used a structure instead. So I defined:
```
typedef struct{
   uint16_t start;
   int16_t  steer;
   int16_t  speed;
   uint16_t checksum;
} SerialCommand;
SerialCommand Command;
```
and then I send the whole structure at once, i.e.:
```
HoverSerial.write((uint8_t *) &Command, sizeof(Command)); 
```
You can see the whole Arduino code here:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/blob/master/02_Arduino/hoverserial/hoverserial.ino

blubbi321, 2020-01-23:
Ok, I tried it out with your fork, everything worked out of the box/repo ;)

Thanks a lot! Also I have to check out all the different variants you implemented, looks very nice!

closed 2020-01-23 by blubbi321