lucysrausch/hoverboard-firmware-hack #130: Unable to use ADC input

Lab-Origami, 2019-11-06:
Hello,

First of all, thanks for your Amazing work.
I would like to make an ebike.
So, i only need to use one motor.
I have this gaz throttle : [Gaz throttle](https://www.amazon.fr/Throttle-Control-Poign%C3%A9e-12V-99V-%C3%89lectrique/dp/B07CZJRK3P/ref=sr_1_1?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=Throttle+Control+Poign%C3%A9e+de+Guidon+et+C%C3%A2ble%2C+12V-99V+Poign%C3%A9e+%C3%89lectrique+Poign%C3%A9e+de+Poign%C3%A9e+Twist+Scooter+Poign%C3%A9e+de+Poign%C3%A9e+Twist+Poign%C3%A9e+de+Poign%C3%A9e+avec+%C3%89cran+LED+Cl%C3%A9+Frapp%C3%A9e&qid=1572951257&sr=8-1) 
I have rebuild the firmware after making changes in config.h
I succesfully flash the board.
But nothing appened.
I know that it is a software issue because I found an already compiled firmware that work like I wanted. But I prefer to really understand what I do, so I try to do all by myself.
Also I'm not able to read any serial data on serial3.
Thx for help

Jan--Henrik, 2020-03-20:
Did you get it to work? If not, please post your config.h and main.c

closed 2020-03-24 by Jan--Henrik

Lab-Origami, 2020-03-25:
Hello,

Sorry for answering so late.
Everything is working great.
The bug was between the screen and the chair as usual 😉
Thanks for your work

Best regards,

Mickael Postolovic

----- Mail d’origine -----

https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/130#issuecomment-601489044