lucysrausch/hoverboard-firmware-hack #139: Flashing Problem ST-LINK

ACG-TR, 2020-03-25:
Hello I am new here, I try to flash the firmware i created from pioneerland to the new hoverboard mainboard i have.

I tried powering the mainboard from 36V battery, from the ST32 dongle(while disconnecting the battery) the red light is blinking when powered but i always get the same error as "cannot connect to target"

the cable connections are correct i checked 10 times already:) GND to GND CLK to CLK and DIO to DIO.

I was not able to unlock the unit as there is no clear tutorial about it. I am using windows 10. is it the problem?

Can anyone help please.

larsmm, 2020-03-25:
I am using Windows 10 without any Problem. Update the firmware of your ST-Link v2 dongle using the newest version of ST-Link utility.

1. connect GND, CLK, DIO, connect USB
2. connect battery
3. press and hold power button of the mainboard
4. flash
5. release power button

ACG-TR, 2020-03-25:
Already did those. St link utility does not see the board on the first page.

larsmm, 2020-03-25:
Try to isolate the error: is it possible to flash another target?

ACG-TR, 2020-03-30:
Hi, I tried your way, i bought bunch of mainboards. 2 of them, same result. But i suceeded in 1 older version. I think in the newer versions some kind of hardware protection is available. 

thanks.

closed 2020-03-30 by ACG-TR