lucysrausch/hoverboard-firmware-hack #140: Wierd sound

DAVID9SE, 2020-03-26:
Hi I know this is not exactly a firmware problem but I don't know what community should I contact so I will post it here. 

So my hoverboard main board works perfectly with the firmware but it always has a weird buzzing noise(not the buzzer) that changes depending on if the motors are on or off or if I spin them manually but never stops.
I think the problem is somewhere in the power supply part but I am not sure. Any idea on how to fix this or if it is even worth fixing if everything works fine.

Thanks a lot.

snackfart, 2020-04-10:
thats coil noise

DAVID9SE, 2020-05-02:
Is that normal with this board or can it be reduced by replacing a component?