lucysrausch/hoverboard-firmware-hack #141: Mainboard dead after 10 minutes

ACG-TR, 2020-04-02:
Hi all, I finally managed to get 1 out of 3 mainboards flashed with the original firmware of Niklasfauth. with PPM setting in config. I had a PWM receiver so i had to use an PWM to PPM encoder to convert the signal to PPM which eventually worked real good. i had steering and forward backward working quite fine. after using the wheels for 10 mins around i stepped on the item that i made (i weigh 80kg) the wheels slowed down as if they were struggling and eventually a beep sound came and i lost power. 

lateron when i checked the power GND and Line i saw that it is shorted out. and after following what is the problem i found out that 1of the MOFSET(Transistor) is burnt. i replaced it with the correct replacement and after 3 second powering up(without wheels connected only board powered) the same transistor again burnt out. 

Anyone faced such issue? Might it related to a different STM pinout? is this a firmware related issue or hardware related badluck? I dont want to waste more money on newboards if this will be the same. 

Wish to receive your comments. Thanks

corteXnzt, 2020-05-02:
Hi! I will be doing PPM setting in config too and I'm not sure if this is related to your question since you have use it for around 10mins as you have said. 

STM32F103 communicates on 3v TTL. Upon reading the datasheet not all GPIO's are 5v tolerant. My guess is that overvoltage on the PPM input pin maybe. The pin where PPM is connected on USART2_RX which is pin PA3, is not 5v tolerant based on the datasheet and most of all receiver and ppm encoder are logic 5v.
[image]
[image]

The debug port which is USART3_TX and RX are 5v tolerant. 
[image]

So this is advisable to be reversed. USART_3 RX should be PPM input. and USART2_TX should be the debug output pin. I'm still looking at the code and currently the things to be change is on config.h and control.c.  Or second option use logic level converter.

There are users who used PPM succesfully but few documentation. And most of them based on pictures, connects the receiver directly to the motherboard signal.

farev, 2020-06-16:
Hi, I have the same problem but I am using a analog joystick to control it. After using it for a while it just stops working it happened to me with another motherboard so I bought another one and I got the same problem.

corteXnzt, 2020-06-16:
You have a picture and schematic (even draft)? Cause that would really help than just by guessing.

farev, 2020-06-16:
Sorry I do not have any schematic.

farev, 2020-06-16:
But I kept testing right now and the wheels started moving without even having the joystick connected, so I will try another firmware to see if that helps.

corteXnzt, 2020-06-16:
Maybe you haven't commented out the automatic testing of motors on power up on the config setting. That is why your motors move?

farev, 2020-06-16:
I did do that, I was abale to use it perfectly for a while and then it stated having this problem. i don't if maybe the input signal from the joystick is causing the probems.

corteXnzt, 2020-06-16:
Yeah maybe, cause the inputs are floating.

farev, 2020-06-17:
It isn't the joystick causing the problem, now it doesn't even move the motors.