lucysrausch/hoverboard-firmware-hack #142: Possible mistake in the reverse engineered schematics?

Forought, 2020-04-11:
Component Q4 either has a wrong name or the diagram drawn may be incorrect.
Please correct me if am wrong.
[image]
Thanks,

corteXnzt, 2020-05-02:
[image]
Orientation is good based on my multimeter check, but resistor connected on my board to emitter is labeled R14.

bahman1920, 2025-01-06:
Hello
The schematic is wrong
I am also building this board
Can we collaborate?
That transistor is G1

bahman1920, 2025-01-06:
[image]