lucysrausch/hoverboard-firmware-hack #143: Hoverboard problem

moezmahroug, 2020-04-30:
Hi,
I'm a new member of this wonderful forum. I would like you to help me find a solution to my problem, please!

So, my problem is that I purchased Hoverboard for robot projects. Unfortunately, I discovered that it is not original. My project is to control their motors with Arduino. So, I tried to work on it, but I made a mistake and I lost the firmware. There are three CPUs and I only work with one of them. I will mention it in the pictures.
If you can help me get him back to work as they were or given me a solution to control it with Arduino for a robot project ( forward, backward, turn left, and right, with variable speed).
I would be very grateful.

type of board => YK95 V1.1

corteXnzt, 2020-05-02:
Check if the pinouts are the same, like the UART ports, LED pin, etc. There is an schematic on the main page of this repo. If all good then you can try uploading the firmware.

GD32F130 can be a replacement for STM32F130 based on this video
https://www.youtube.com/watch?v=xej19edPU9M

Or another option buy new board similar to the one in this repository.

moezmahroug, 2020-05-02:
Unfortunately, is not the same, :/ 

[image]

moezmahroug, 2020-05-02:
and I think I will try to eliminate the original circuit and replace him with Arduino directly. 

[image]

[image]

jagimene, 2020-05-03:
Hi, what hoverboard did you bought. I have an electric skateboard and I want to mod it. The motherboard have the same cpu gd32f103rct6

FIRSTQWERT, 2022-05-28:
> Hi, what hoverboard did you bought. I have an electric skateboard and I want to mod it. The motherboard have similar cpu gd32f103rct6. So I searching for a way to reprogram it

Hi, did you find solution to reprogram gd32f103rct6? I have same problem.

Luisba1958, 2022-08-09:
Is posible that one gyro sensor is an empty board? The botom one in one picture