lucysrausch/hoverboard-firmware-hack #146: PPM/PWM Wiring to Receiver?

mkunz7, 2020-06-07:
I'm not seeing any documentation on how to connect the three wires from the receiver to the motherboard. Can someone help me out?

I was watching https://www.youtube.com/watch?v=Ww_59NLo0_o and noticed on snowkingcros PWM build he claims to use ADC1 (ch1 steering) and ADC2 (ch2 throttle) on the left sensor board. He references this screenshot. He flashes benjaf's firmware. Is this the correct way to wire for ppm? 
[image]

benjaf, 2020-06-09:
The screenshots above show only one channel connected (Green), the other wires are +5V and GND.
I have had it connected the same way. Make sure to use the 5V line, NOT the 15V one.
In theory the output should be shifted to 3.3V, but I seem to recall those particular pins being 5V tolerant.

k3nden, 2020-06-09:
benjaf, I can make both wheels respond using the green wire (green signal, black gnd). I get nothing on the blue wire (blue signal, blk gnd). Can you help me locate the code segments that I need to modify to make it drive like a tank? I was compiled your "esk8" build. TIA

mkunz7, 2020-06-09:
Excuse my lack of knowledge, but I'm confused how this would work. Channel 2 only interfaces to the throttle, not steering (ch1). Also most receivers need more than 3.3v to operate? The manual for a spektrum sr315 says 3.5-9.6v.

benjaf, 2020-06-10:
@mkunz7 Yes, channel 1 is steering and channel 2 is throttle. In the screenshots above, only throttle is connected. In theory the signal from the receiver to the mainboard (blue / green wire) should be dropped to 3.3V with a "Level converter" to ensure the STM32 is not damaged. This does not seem to strictly necessary, but may cause long term damage to the STM32.
@k3nden If you are using the esk8 build, steering is not available by default!

benjaf, 2020-06-11:
@k3nden I have no idea what would cause that, the input should reset to safe values if valid input is lost. Be aware that the signal lines are highly sensitive to electronic noise. Long wires close to the switching MOSFETs will cause problems, especially at higher speeds.

k3nden, 2020-07-18:
> [working.mp4.zip](https://github.com/NiklasFauth/hoverboard-firmware-hack/files/4761019/working.mp4.zip)
> 
> Works great thanks for the help guys!

@mkunz7  How have you mounted the motors? Did you use a pipe as you did for the skateboard? I would appreciate it if you could provide a few details and closeups. TIA.