lucysrausch/hoverboard-firmware-hack #149: Reverse-engineered schematics pdf link is down.

FirePT, 2020-06-17:
I was looking to find one of the  smd capacitor values/part number and found out that the link to the schematics is not working.
If anyone can help me out with my search it would be great since I was de-soldering one of the mosfets that was faulty and the heat of the soldering iron make the smd capacitor (0402 brown one very close to each mosfet) to also de-solder and I lost it and have no idea what type it is...

FirePT, 2020-06-18:
[image]

I think that with the picture is easier to explain.

k3nden, 2020-06-19:
@FirePT - Look here. https://github.com/NiklasFauth/hoverboard-firmware-hack/issues/147

FirePT, 2020-06-19:
Amazing @k3nden! Thank you 👍 

The condenser I need is not specified... have to continue searching or have to desolder another one from the pcb and have it measured.