lucysrausch/hoverboard-firmware-hack #152: LENZOD Board - Error during firmware update

marco-teixeira, 2020-06-23:
Hello everyone!

I bought a hoverboard used to try to convert to a cart controlled by a raspberry Pi via serial communication. The principle is not to give problems.

However, I am unable to modify the firmware. I connect GND, SWDIO, and SWCLK to ST-Link v2 and the battery. When I turn on the board, it beeps continuously, and the computer only shows the message "no device connected". The application recognizes St-Link, but does not recognize the board.

I tried to follow some tutorials, which say that you need to unlock the board first, but the same error always happens. Has anyone been through something similar? Could anyone suggest a solution?

Thank you all!

[image]
[image]

marco-teixeira, 2020-06-25:
[image]
[image]

FirePT, 2020-06-29:
Hi @marco-teixeira , for having better chances of someone to help take another close up photo just at the "ARM" chip so it can be identified.

marco-teixeira, 2020-06-29:
Hello @FirePT , here is a picture of ARM. Apparently it is the same found in the photos of the tutorial. Maybe the debug pins are not connected? But it is strange, because when I connect to the St-link, the sign will beep continuously.

[image]

corteXnzt, 2020-07-18:
Use a continuity meter and check if your pin connections are correct. Check the datasheet of the chip and identify gnd, swdio and sclk. Try also opening the stlink programmer cause sometimes the labels on aluminum case are not what on the PCB like on my programmer.

bekirbostanci, 2020-07-30:
I had same card (lenzo) and same problem I did not solved i tried many things but didnt solve the problem. You can buy a new card in aliexpress

adverst, 2020-10-02:
Hey I have this board too and initially had the same problem... But, pin order is different to the one in the images for the repo.
3.3v is the same as the board in the image, but then it is SWCLK, SWDIO, GND. If you get the pin order right this board flashes ok.

lekse, 2021-06-26:
A same board and same problem. So I checked datasheet and draw picture for someone else, who will have a same problem with Lenzod board.

[image]

angelo81523, 2022-05-01:
You should see which capacitor or resistor to dissolder from board to access the "download mode" @lekse

shivamlekhi, 2025-02-28:
> A same board and the same problem. So I checked datasheet and draw the image. And I leave it here for someone else, who will have the same problem with Lenzod board.
> 
> [image]

These are the correct pins, Lifesaver! Thanks much!