lucysrausch/hoverboard-firmware-hack #153: Wii Nunchuck no reverse

patrickja, 2020-07-29:
I am currently working on setting up the hoverboard with the nunchuck option, but I am running into an issue. I have gone through the issues logs for this firmware, as well as the firmware issues and completed for a number of offshoots of this firmware. I have triple checked my wiring and I know it is correct by the fact that I have some control of the board. I have attempted many of the code "fixes" for similar issues with no luck. I am also using the legit Nintendo brand nunchuck. 

The code I am using is a fresh copy of this firmware with only the change to /Inc/config.h in which I uncomment the define for nunchuck control input and comment out the debug for UART3. I receive no errors when I do a make and no errors when I do an upload to the board. 

The issue is that I can only drive the system in forward and left and right. If I try to reverse I get a program shut down, which forces me to hard restart the board. I have tried numerous nunchucks as well to rule out the nunchuck. 

Any suggestions would be much appreciated. 

Thank you.

vamfun, 2020-12-30:
Did you solve this?  I have a similar problem with my HOVER - 1 H1 model motherboard.  Any command  that transitions to reverse direction either + to - or vice versa causes a shutdown that requires a power on restart.  If the board starts with a command you can adjust the motors with ADC but disconnect when command gets below 10.   
Any thoughts?

sdegeorgio, 2022-03-06:
Hi, I had the same problem looking for a repo that worked well with nunchuck support. In the end I used the EFeru hoverboard firmware hack fork and fixed the nunchuck driver to make it more robust. Have a try with that one?
Steph.

Jodar5, 2024-05-07:
hello,
 it's been a good month since I tried to motorize a wheelchair for my mother who is disabled with a nunchuck control.
 I can't get through it. If you could share a valid firmware I would be happiest. 
 Thanks to those who can help me
Jose