lucysrausch/hoverboard-firmware-hack #154: Overvolting the board?

annoyinganime, 2020-09-01:
Does anybody know, what maximum theoretical voltage could those boards run from? Looking at MOSFETs datasheet, those are rated for 75v, and BJTs used in 15v supply are rated for 100v. So can I run the board from 12s lipo? 50.4 volts should not kill it, but maybe I've overlooked something?

o0Mouse0o, 2020-09-14:
It's probably not much help but I've blown either Q2 or Q3, TIP127 when in regen mode on a freshly charged battery and the BMS has gone open circuit to protect the battery cells. ie a massive voltage spike from the board being powered from the regen voltage and having no battery to sink the power into. 
So I guess it might indicate this is the first point of failure with a massive over-volt. 
Kind of related...It's common in the world of ebike modding to change the voltage of the main capacitor so they can be run at the voltage of the main MOSFETS which are then the voltage limiting components.

annoyinganime, 2020-09-14:
So basically I just need to add a TVS diode? (Or not use a BMS, lol)