lucysrausch/hoverboard-firmware-hack #158: Can control Hoverboard motors with PWM but not PPM

mccgregory, 2020-10-13:
Hi, I have been following you for some time. Absolutely excellent work!
Thank you so much for your efforts and your generosity in sharing it- what a great team effort!
In trying to replicate your work – I would like to achieve a 4WD robot driven on ‘missions’ by a drone Flight Controller.
The Flight Controller requires PPM- output.
With your project I can achieve PWM output, but haven’t been able to get PPM to work.
The RC Transmitter I am using is a FlySky FS-i6X Transmitter and FS-IA6B Receiver.
My MainBoard was bought, new,  especially for this project and is very similar to the one shown in NiklasFauth’s GitHub page.
I have checked the output from the Receiver is PPM (as the TX is capable of PWM & PPM) and that it is actually working by connecting the RX to the Flight Controller and checking the virtual controls (also checked with a logic analyser).
Can you think of any diagnosis, Trouble-shooting or fault finding I can do with the Hoverboard Mainboard to help me resolve this?
I do have some background in electronics with the required test equipment – what should I be looking for?
I have no problems with changing the Config.h file and re-compiling – I have tried several different downloads and have EVEN used the code via https://pionierland.de/hoverhack/ and used RoboDurden’s hex/bin files direct without any success for PPM control.

I haven’t tried DEBUG MODE – would this give any indication of the PPM signals being received?

Thank you in anticipation
All the Best with your imaginative projects!
Greg

Candas1, 2020-10-13:
Hi, have you tried changing the number of channels in the firmware ?

mccgregory, 2020-10-14:
Hi, thanks for the reply.

Do you mean in the line:

	

#define PPM_NUM_CHANNELS 6          // total number of PPM channels to receive, even if they are not used.

 

If so, then my Receiver (FS-IA6B) has 6 Channels and therefore the line is correct?

 

Best regards

Greg

Candas1, 2020-10-14:
Try 8

mccgregory, 2020-10-14:
Well that’s puzzling!

It works!

Thank you very much!

Much appreciated – I’ve been struggling with this on-and-off for hours over the last few weeks.

One question here, and I have a solution.

Thanks again!

Greg

mccgregory, 2020-10-14:
**Candas1** suggested I set it to 8 Channels instead of the default (and correct number) of 6.
That works!
Thank you.

closed 2020-10-14 by mccgregory

mccgregory, 2020-10-19:
I have since rechecked the Receiver spec. and everything I have seen shows it as 6 Channels.
I have tested the RX and found although it has 6 x PWM Channels (limited by physical hardware) it is outputting 8 x PPM Channels when set to PPM.

Candas1, 2020-10-19:
Yes I know it because I have the same receiver.
If you want to benefit from FOC motor control and ibus protocol you could also try this firmware https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC
It has a recent bug that has to be fixed though.
Join the telegram group, you can get quick help there as well.

mccgregory, 2020-10-19:
Can't seem to attach file or picture. Previous comment (above) gave me the option but didn't deliver - anyway it shows definite 8 Channels, so the spec. for the RX would be more accurate if it said 6 x PWM Channels.
As a touch of irony, having solved this problem, I don't need to worry as the Flight Controller input is PPM, but the Motor Outputs are PWM! 
:o)