lucysrausch/hoverboard-firmware-hack #162: Motor Command zero crossing causes disconnect

vamfun, 2020-12-30:
I have a strange problem with my HOVER - 1 H1 model motherboard. (See link pictures   https://vamfun.wordpress.com/2020/12/23/flying-firehose-blog-1-hover-1-h1-hoverboard-tear-down/) 
 Any command that transitions to reverse direction either + to - or vice versa causes a shutdown that requires a power on restart. If the board starts with a command you can adjust the motors with ADC but disconnects when command gets below 8.    I’ve tried most Foc firmware and other firmware variants and all produce the same shutdown.   Running a straight motor test download runs a one sided test...motors disconnect on the reverse transition.   
Any thoughts?

DanielQuintana2, 2021-01-09:
Hi @vamfun 
I'm having the same problem here. Have you found any solutions?

vamfun, 2021-01-09:
Yes.... the board has six pins that are wired differently... the problem was the v_battery is not showing any voltage so the board disengages .  Software interlocks keep it from shutting down until the command is low.   That’s why it will allow a half cycle of test run.
Fixing the GPIO cures the problem. 
See issue #126 of EmanualFeru-hoverboard-hack-FOC. “Charging enables motors on Alternate H1 motherboard ...”.  It has the new pin outs.    I am using the FOC firmware since I’m making the motors into servos and need good slow speed performance.    I’ve got the board working using the low resolution hall sensors for position.   Not bad..gets 4 deg resolution 😎

http://vamfun.wordpress.com
Robot musing blog
iPhone

DanielQuintana2, 2021-01-09:
Ah Thank you @vamfun 
I'm looking at issue #126 of EmanualFeru's and I don't see any pinouts...

[image]

Is there any way you could provide me with them or direct me to a diagram where I can see the configuration? Sorry to bother but thank you.

vamfun, 2021-01-09:
I edited my first comment and added some pictures.    Here are changes made to the EmanualFeru FOC fork by @Chaos99 for what he calls Gen3 board.  These are the same changes I made.   [image]image](https://user-images.githubusercontent.com/4359767/104109557-02bc7d00-5284-11eb-96e2-6aa797c1e1a7.jpeg) 

Not sure what happened to #126 issue I posted..seems to be gone.

DanielQuintana2, 2021-01-11:
Thank you so much @vamfun !!!