lucysrausch/hoverboard-firmware-hack #163: Control methods using Serial and limiting current under fast changes

papaxiclas, 2021-01-26:
I there!
I have been using your code for a few months on a loading cart controlling the cart via serial with the setPWM method.
For now it works but there are times I get limited use because of the fast changes in speed.
For example, if I am at full pwm and drop to zero the wheels brake hard enough for the cart to jump out of my bench. If I do the reverse and accelerate from 0 to full power the same "impact" happens. What this does is that when the cart is loaded or on an incline the power of this "jumps" are enough for the BMS and even the board it self to shutdown for overcurrent. 
Is there any other control method I can use the set's the correct PWM dynamically so that the electrical current never goes over a determined value?
I am using the `void setHoverboardPWM( int32_t pwm1, int32_t pwm2 )` function to send the control values to the board.

Thanks

Candas1, 2021-01-27:
Adjusting the filter value you could smoothen the input.

Have you tried this firmware:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

It is more power efficient as it uses FOC control, and has a maximum current setting that could help.

Are you sure BMS shuts down because of the overcurrent protection ? 
If your battery is bad the voltage drop when drawing high current could also lead the software to shut the [board off](https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC/wiki/Battery).

papaxiclas, 2021-01-27:
I will give it a try but the problem is the serial interface. I tried a few forks a few months ago but the serial protocol was not fail proof and I got the cart going full throttle without control a few times.
About the BMS, either way is valid. The voltage drops if the current is high enough so if it cuts from high current or low voltage leads to the same original issue. But I am convinced it is current because the behavior is the same regardless of charge level on the battery.

larsmm, 2021-05-21:
This Project is deprecated. There is an more advanced version that has emerged from this project:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC