lucysrausch/hoverboard-firmware-hack #166: GD32F103 RCT6 CE6W495 AC1702 What should I do with a board equipped with?

doraorion, 2021-02-04:
GD32F103 RCT6 CE6W495 AC1702
What should I do with a board equipped with?

Candas1, 2021-02-05:
Do you have a picture of your board ?
GD32F103RCT6 chip is ok.

doraorion, 2021-02-05:
[image]
[image]

doraorion, 2021-02-05:
It is a picture of the board

Candas1, 2021-02-05:
Are the cables having 4-5-5-4 wires ?
I see some wires that have a different color than usually but otherwise it looks ok.
Have you replaced a TIP127 ? was is initially facing down like that ? is there another one under the board ? 

I had started to collect different models of hoverboard mainboards:
https://docs.google.com/spreadsheets/d/1Vs3t2W8_z7E93Ij0pwx_cKzxzKSvjW1n1t_40CXV3ik/edit#gid=0

doraorion, 2021-02-05:
Yes, the cable is 4-5-5-4 wire.
What is TIP127? Not exchanged

Will those models behave the same?

Candas1, 2021-02-05:
It's that composant that is twisted. 
It looks like it burned

doraorion, 2021-02-10:
Replaced TIP127
However, although I can write from my computer, I cannot turn on the power when the battery is connected.

Candas1, 2021-02-15:
Sorry for the late reply.
I think that's the problem, the TIP127 turns the power on.
Are you sure you replaced it properly ?

larsmm, 2021-05-21:
This Project is deprecated. There is an more advanced version that has emerged from this project:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC