lucysrausch/hoverboard-firmware-hack #169: yst dxt j20  v5.5

Rettigran, 2021-05-07:
hi
i am building one wheels, connecting both wheels toghter to be controlled by same dougther board . 
before i used the ARM  controllers , and connected the 2 dougther  bords  together by connecting the green and yellow wires on both 4wire connectors.
i just got this new controller , and could not find the wiring diagram.
does anybody has one? has anybody connected this type in  this way?

larsmm, 2021-05-21:
This Project is deprecated. There is an more advanced version that has emerged from this project:
https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC

Rettigran, 2021-05-24:
Thanks 

 

 

רטיג רן

 

גנון טכנולוגיות בע"מ.

מקלף 41, מפרץ חיפה.

טלפון: 04-8417843

פקס:   04-8419330  

נייד:  052-2818590 

 ***@***.***> ***@***.***

 

 <http://www.ganontech.co.il/> www.ganontech.co.il

 

https://github.com/EmanuelFeru/hoverboard-firmware-hack-FOC