lucysrausch/hoverboard-firmware-hack #171: make: *** [Makefile:145: build/hover.elf] Error 1

Crowerade, 2021-05-14:
Hello,
I am on Manjaro and I'm trying to compile the hoverboard-firmware-hack. However, each time in enter 'sudo make' inside the folder, the terminal returns to me this error : 
```
collect2: error: ld returned 1 exit status
make: *** [Makefile:145: build/hover.elf] Error 1
```

Brnpimenta, 2021-09-11:
hey crowerade,

I got the same error:
https://github.com/cloidnerux/hoverboard-firmware-hack/issues/3

any insght to solve it?

regards,

Bruno Oliveira

w-A-L-L-e, 2022-09-17:
@jonescb5 Indeed same issue here, when compiling today on macOS had to make following patch to get it to compile:
```
diff --git a/Src/comms.c b/Src/comms.c
index 5e437a1..5e5c323 100644
--- a/Src/comms.c
+++ b/Src/comms.c
@@ -5,7 +5,7 @@
 #include "stdio.h"
 #include "string.h"
 
-UART_HandleTypeDef huart2;
+extern UART_HandleTypeDef huart2;
 
 #ifdef DEBUG_SERIAL_USART3
 #define UART_DMA_CHANNEL DMA1_Channel2
diff --git a/Src/control.c b/Src/control.c
index 9ffdfd2..06247a2 100644
--- a/Src/control.c
+++ b/Src/control.c
@@ -14,8 +14,8 @@ uint8_t nunchuck_data[6] = {0};
 uint8_t i2cBuffer[2];
 
 extern I2C_HandleTypeDef hi2c2;
-DMA_HandleTypeDef hdma_i2c2_rx;
-DMA_HandleTypeDef hdma_i2c2_tx;
+extern DMA_HandleTypeDef hdma_i2c2_rx;
+extern DMA_HandleTypeDef hdma_i2c2_tx;
 
 #ifdef CONTROL_PPM
```

After that I get a has file in build/hover.hex (also some warnings on readr.c and writer.c from libc/reent).

GhajariAli, 2023-07-16:
I did add extern to those three but still the same error. any idea on the issue? 
also there are lots of warnings...

GhajariAli, 2023-07-16:
ok looks like it's nagging about this in the bldc.c but I don't understand what's wrong!
undefined reference to `blockPWM'

youcefK, 2025-07-23:
Here us the solution
Chage bldc.c file ithe the originale firmware file