lucysrausch/hoverboard-firmware-hack #172: at32f403 using st link, openocd, linux not recognizing.

Brnpimenta, 2021-07-20:
Hello,
I have a board which uses at32f403 chip, I'm trying to use st link, openocd and linux, but somehow it's not recognizing.

Can you share a step by step to it?

Candas1, 2021-07-27:
https://github.com/cloidnerux/hoverboard-firmware-hack

zuingo, 2022-07-29:
Thanks i will try!  I have always less time!

zuingo, 2022-08-07:
Hi, i tried with lubuntu and the three openocd's command present in the giude...

[image]

but i can't remove RDP.

[image]

[image]
 i see the warning

[image]

but...i'm not a linux users so i don't know if i can do something else or i'm doing something wrong...any other that i can try? thanks in advance it can be usefull to same others

zuingo, 2022-08-07:
i try to remove -v2 from openocd command but nothing..
[image]
.

Candas1, 2022-08-08:
I have 0 experience with this. I suggest you ask the question in that repository

Candas1, 2022-08-09:
You might also want to show your interest [here](https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/329)
But don't post your problem there lol