lucysrausch/hoverboard-firmware-hack #182: yst dxt j20 v5.7

zuingo, 2022-07-25:
Hi to all,
i buied two main boards yst dxt j20 v5.7 with AT32F413RCT7 to make a 4x4 chair for a disabled friend.
My problem is that i'm not able to remove read only protetction with stm32cubeprogrammer or stm32 link utility.
Can someone give me an hand?
Thanks in advance
[image]
[image]
[image]
[image]

Candas1, 2022-07-25:
Hi,

You should read the other issues, it will help

zuingo, 2022-07-25:
Hi and Thanks,
i read and i try all that i can read and find on net for at32403, i try with jlink but i don't have jlink usb device, i try with Keil uVision5 but nothing...i try to put nrst 7 pin  in ground...at me seams that nothing work...
There is someting that i can read better?

[image]

Candas1, 2022-07-25:
https://github.com/lucysrausch/hoverboard-firmware-hack/issues/172

zuingo, 2022-08-07:
i try to remove -v2 from openocd commad but nothing...i don't see the warning
[image]