lucysrausch/hoverboard-firmware-hack #183: Extract one channel.

Kambodia91, 2023-01-03:
hello, I have a hover board with two separate bldc control boards, I soldered stm32f103CBT6 48pin on them. Is it possible to extract the foc control for the bldc controller, e.g. via usart? The circuit is equipped with current measurement on two phases, voltage measurement and main current measurement. I also have a pinout of how the stm32 is connected
[image]
[image]
both boards are connected by usart0 PA2 and PA3.

Candas1, 2023-01-03:
https://github.com/EFeru/hoverboard-firmware-hack-FOC

Kambodia91, 2023-01-04:
I know about you. but this software is not compatible with my stm32 version. in platform.ini the stm32f103RC processor is declared, it is a processor with 64 pin outputs. I have stm32f103CB in 48pin outputs. Is there anyone who can extract the code for only one motor / controller? Regards.

Kambodia91, 2023-04-29:
Model informatorom: https://github.com/EFeru/hoverboard-firmware-hack-FOC/issues/364

closed 2023-04-29 by Kambodia91